ros-drivers/phidgets_drivers #4: Interface Kit Support

sevenbitbyte, 2017-03-02:
I supported the Phidgets Interface Kit a while ago but due to the long awaited release of this repo for jade/kinetic have not released it as a seperate package. If you would like I'd happily submit a PR here for to support the interface kit directly to this repo as I think this probably a better place for it.

https://github.com/CumulonimbusLtd/phidgets_interface_kit

mintar, 2017-03-05:
Hi @sevenbitbyte , thanks for your offer! I've just browsed over your repo, and I'd definitely like to include it in `phidgets_drivers`. I don't have an Interface Kit for testing, so I'd have to rely on you to ensure it's working (and not just compiling). Also, it would be great if we could reduce code duplication as much as possible. I haven't reviewed it all, but for example `Phidget::registerHandlers()` from `phidgets_api` could be reused in you code; conversely, I guess some of your functions could be moved to `phidgets_api` for consistency. Having consistency would help me out a lot while maintaining this stuff later, especially since I don't have a physical device for testing. If you're willing to put that work in, please go ahead an open a PR, I'd really appreciate that!

mintar, 2017-05-22:
Closing due to inactivity, feel free to reopen if you are still interested.

closed 2017-05-22 by mintar