ros-drivers/phidgets_drivers #5: Motor driver with odometry

sevenbitbyte, 2017-03-02:
I've implemented a driver which provides basic move_base support using phidgets encoders to produce odometry and uses the phidgets HC motor controller for motion control. This probably needs some work but if you would like to borrow some of this code or think a PR to this repo is the right place for this I'm happy to help get this brought over.

https://github.com/CumulonimbusLtd/phidgets_motion_control

mintar, 2017-03-10:
Same as with #4 : This looks like a great addition, if you're willing to work on a PR. Some of the code would have to move to `phidgets_api`, and can probably use the new files added in #3 once that's merged.

mintar, 2017-03-10:
Also related: https://github.com/ccny-ros-pkg/phidgets_drivers/pull/23

mintar, 2017-05-22:
Closing due to inactivity, feel free to reopen if you are still interested.

closed 2017-05-22 by mintar