ros-drivers/phidgets_drivers #6: PhidgetsImuNodelet is not getting parameters from launch file

ItsJas, 2017-04-24:
I was trying to set the compass parameters in "imu.launch" but I'm getting "No compass correction params found."

mintar, 2017-04-24:
Can you please copy-paste the contents of your launch file?

ItsJas, 2017-04-24:
<launch>

  #### Nodelet manager ######################################################

  <node pkg="nodelet" type="nodelet" name="imu_manager" 
    args="manager" output="screen" />

  #### IMU Driver ###########################################################

  <node pkg="nodelet" type="nodelet" name="PhidgetsImuNodelet" 
    args="load phidgets_imu/PhidgetsImuNodelet imu_manager" 
    output="screen">

    # supported data rates: 4 8 16 24 32 40 ... 1000 (in ms)
    <param name="period" value="4"/>

    # compass correction params (see http://www.phidgets.com/docs/1044_User_Guide)
         
         <param name="cc_mag_field" value="0,49966"/>
         <param name="cc_offset0" value="0,04192"/>
         <param name="cc_offset1" value="0,24596"/>
         <param name="cc_offset2" value="-0,02791"/>
         <param name="cc_gain0" value="1,94740"/>
         <param name="cc_gain1" value="1,91920"/>
         <param name="cc_gain2" value="2,13750"/>
         <param name="cc_t0" value="-0,02415"/>
         <param name="cc_t1" value="-0,05631"/>
         <param name="cc_t2" value="-0,02452 "/>
         <param name="cc_t3" value="-0,02640"/>
         <param name="cc_t4" value="-0,06142"/>
         <param name="cc_t5" value="-0,02846"/>  
  #### IMU Orientation Filter ###############################################
  <node pkg="nodelet" type="nodelet" name="ImuFilterNodelet" 
    args="load imu_filter_madgwick/ImuFilterNodelet imu_manager" 
    output="screen">
    <param name="use_mag" value="true"/>
    <param name="publish_tf" value="false"/>
  </node>
</launch>

ItsJas, 2017-04-24:
It was because of the " ," instead of "." , What a careless mistake !  Thank you for you quick answer as usual.

closed 2017-04-24 by ItsJas