ros-drivers/phidgets_drivers #17: Out-of-order packets

alexwhittemore, 2017-10-11:
Here's another one I hope to characterize better tomorrow, but I want to get it written in case anyone's seen this behavior before: it appears sometimes that the driver publishes IMU messages to the topic out-of-order in time. That is, a message will hit the topic with an earlier capture time than the most recent, as if there's a race condition somewhere in the pipeline ahead of publishing. In most applications, this probably doesn't matter much, but for whatever reason, Cartographer chooses to handle it with an instant choke-and-die.

mintar, 2017-10-11:
That sounds like a bug. If you can reproduce it, it would be great to have a rosbag of this (also for the other issues).

alexwhittemore, 2017-10-11:
That I can do! I'll get one today.

commit a5523f34 (2018-08-07): phidgets_imu: Ensure strictly ordered timestamps (#26)

Fixes #17.
closed 2018-08-07 by mintar