ros-drivers/phidgets_drivers #18: Rotational drifting

alexwhittemore, 2017-10-11:
The last not-fully-characterized issue I'll post tonight, I promise. Every so often, our robot appears to be freely rotating about Z in space. It's unclear what the root cause of the issue is, but when it comes up, power cycling the IMU seems to clear it up, at least temporarily. This suggests to me it's not actually a driver issue, but has anyone else seen gyro offset like this? Could it be related to magnetometer calibration (or lack thereof)?

mintar, 2017-10-11:
> The last not-fully-characterized issue I'll post tonight, I promise.

Keep 'em coming, I'm happy to help. :)

To be honest, I haven't had any good results with the magnetometer on a ground-based indoor robot. The electric motors induce a magnetic field that's orders of magnitude larger than the earth magnetic field, and since it's a dynamic field, it can't be calibrated out. Also other metal parts in the environment introduce disturbances. I always turn off the magnetometer in the orientation estimation (of the imu_filter_madgwick package).

Also, yes, gyros do drift. The drift is calibrated out during startup (so it's important that the IMU is at rest when starting the driver):

https://github.com/ros-drivers/phidgets_drivers/blob/8b22628743a143cede8125fe7842304116573785/phidgets_imu/src/imu_ros_i.cpp#L170

However, drift accelerates as the IMU heats up, so usually you'll get a systematic drift after 30 minutes or so of operation. What you can do is call the imu/calibrate service every once in a while when the robot is at rest. It isn't done automatically in the driver, since the driver can't know when the robot is at rest.

alexwhittemore, 2017-10-11:
Oh that's a great tip, certainly a lot easier than power cycling the whole shebang. I'll try it, thanks!

closed 2018-01-17 by mintar