ros-drivers/phidgets_drivers #21: Connect two spatial

matteopantano, 2018-01-04:
Hi,
 
I am trying to use phidgets_imu. I have two spatial with different serial numbers but when I try to run the launch file provided it always connect to the same one. I am not able to create two launch files that launch two different sensors, I have changed the name of topic s but still does not work!

```
<!-- Phidgets IMU launch file -->

<launch>

  #### Nodelet manager ######################################################

  <node pkg="nodelet" type="nodelet" name="imu_manager" args="manager" output="screen">

  </node>

  #### IMU Driver ###########################################################

  <node pkg="nodelet" type="nodelet" name="PhidgetsImuNodelet" args="load phidgets_imu/PhidgetsImuNodelet imu_manager" output="screen">

    <remap from="/imu/data_raw" to="/imu_top/data_raw"/>
    <remap from="/imu/mag" to="/imu_top/mag"/>
    <remap from="/imu/is_calibrated" to="/imu_top/is_calibrated"/>

    # supported data rates: 4 8 16 24 32 40 ... 1000 (in ms)
    <param name="period" value="4"/>
    <param name="frame_id" value="imu_link_top"/>

    # optional param serial_number, default is -1
    <param name="serial_number" value="301940"/>

    # compass correction params (see http://www.phidgets.com/docs/1044_User_Guide)
    <param name="cc_mag_field" value="1.04193"/>
    <param name="cc_offset0" value="0.01966"/>
    <param name="cc_offset1" value="0.12772"/>
    <param name="cc_offset2" value="0.00000"/>
    <param name="cc_gain0" value="0.98090"/>
    <param name="cc_gain1" value="0.93862"/>
    <param name="cc_gain2" value="0.95976"/>
    <param name="cc_t0" value="0.00346"/>
    <param name="cc_t1" value="0.00000"/>
    <param name="cc_t2" value="0.00361"/>
    <param name="cc_t3" value="0.00000"/>
    <param name="cc_t4" value="0.00000"/>
    <param name="cc_t5" value="0.00000"/>
  </node>

  #### IMU Orientation Filter ###############################################

  <node pkg="nodelet" type="nodelet" name="ImuFilterNodelet" args="load imu_filter_madgwick/ImuFilterNodelet imu_manager" output="screen">

    <remap from="/imu/data_raw" to="/imu_top/data_raw"/>
    <remap from="/imu/mag" to="/imu_top/mag"/>
    <remap from="/imu/data" to="/imu_top/data"/>

    <param name="use_mag" value="false"/>
    <param name="use_magnetic_field_msg" value="true"/>
    <param name="world_frame" value="enu"/>
    <param name="publish_tf" value="false"/>
  </node>

</launch>
```
This is the example of my launch file, I tried to insert the param serial_number but no results.

mintar, 2018-01-11:
Hmm, strange. Unfortunately I only have one Phidgets IMU available for testing. I tried the following:

* don't set `serial_number` param -> connects to first device
* set `serial_number` param to `-1` -> connects to first device
* set `serial_number` param to correct number -> connects to specified device
* set `serial_number` param to an invalid number -> doesn't connect to any device

So that seems to be correct, and I would have hoped that it works with two devices. Can you attach some more log output that demonstrates the problem?

mintar, 2018-08-07:
Closing due to inactivity.

commit cbd1448e (2016-11-25): Update README (specific ROS versions)

Fixes #21.
closed 2018-08-07 by mintar