ros-drivers/phidgets_drivers #23: Services' type

dogoepp, 2018-01-23:
Hello,

I realise now that my approach to services for IL, in #22 might not be the best.
Wouldn't it be better to define a single service with two parameters : id of the output (int) and state (bool) ?

mintar, 2018-01-23:
Hmmm... both approaches have their merits.

Single service:

* simplifies the calling code: you only need one service client if you want to set multiple outputs

One service per output:

* you can remap the service name to something semantically meaningful
* doesn't need a custom service definition

But overall I agree that having a single service would probably be better. If you want to do the change, let's do it now before I make a new release; I try to keep the API stable, but since we only introduced these services recently I think nobody (except you) is already using them.

dogoepp, 2018-01-24:
I agree with you. Here is one proposal (#24). However, it allows for only one output to be set at a time. I can change it so that any number of outputs can be "simultaneously" set.

commit f61f96f5 (2018-01-25): Move from one service per output to a single one (#24)

Tested with the 0/0/4 Phidgets InterfaceKit.
Mistakenly, the same service was advertised as many times as there are outputs. This was fixed.
Closes #23.
closed 2018-01-25 by mintar