ros-drivers/phidgets_drivers #25: phidgets_ir?

Shantanu-Karnwal, 2018-03-11:
I don't see any folder for phidgets_ir in this repo. Wheras it was there in the earlier repo - https://github.com/ccny-ros-pkg/phidgets_drivers. Also it is mentioned here - http://wiki.ros.org/phidgets_drivers?distro=indigo (btw I am using kinetic). What happened?

mintar, 2018-03-12:
I removed that package in e1ed135, when first releasing phidgets_drivers to kinetic; i.e., it's still in indigo, but not kinetic. The reason for this is that phidgets_ir was only a stub; it didn't publish any topics, it only printed the IR output to the console. The support for IR is still in the phidgets_api package (https://github.com/ros-drivers/phidgets_drivers/blob/93a60996eebb4d77ad97b86c13762bca8744d53a/phidgets_api/src/ir.cpp), but there is no ROS wrapper for it.

If you would like to see support for the IR sensor, please feel free to open a pull request. I would suggest to do a `git revert e1ed135` and then implement the `IRRosI::codeHandler()` method so that the IR readings are published as [`sensor_msgs/Range`](http://docs.ros.org/api/sensor_msgs/html/msg/Range.html) messages. I don't have a Phidgets IR device for testing, so I can't do it myself, but I would be glad to review and merge a pull request.

Shantanu-Karnwal, 2018-03-12:
Well, if the phidgets_api is all the ir needs, then that's fine. If it is necessary, I'll definitely write one package for the phidgets_ir. Thanks a lot!

closed 2018-03-12 by Shantanu-Karnwal