ros-drivers/phidgets_drivers #29: Garbage diagnostics data when the IMU is not connected

jspricke, 2018-11-05:
We fill hardware IDs if the IMU is not connected here:

https://github.com/ros-drivers/phidgets_drivers/blob/kinetic/phidgets_imu/src/imu_ros_i.cpp#L174

This leads to broken strings (we added a `ROS_FATAL` below to print the data):
``
[/broken/imu_manager  INFO 1541437152.313089147]: Starting Phidgets IMU
[/broken/imu_manager  INFO 1541437152.349699310]: Opening device
[/broken/imu_manager  INFO 1541437152.349842384]: Waiting for IMU to be attached...
[/broken/imu_manager FATAL 1541437162.359115469]: Problem waiting for IMU attachment: Given timeout has been exceeded. Make sure the USB cable is connected and you have executed the
phidgets_api/share/setup-udev.sh script.
[/broken/imu_manager  INFO 1541437162.359166272]: Calibrating IMU...
[/broken/imu_manager  INFO 1541437162.359194949]: Calibrating IMU done.
[/broken/imu_manager FATAL 1541437162.359280129]: h?W?_ -32767
[/broken/imu_manager  INFO 1541437162.359303531]: No compass correction params found.
``
Also the `calibrate()` in the function probably only makes sense if the IMU is connected. So maybe adding both to an `else` case is enough?

mintar, 2018-11-06:
Sounds good. We should probably also set the hardware ID to "NONE" or "" or whatever in the other case (when the IMU isn't connected). And while we're at it, we should also change the tabs to spaces in that function.

PR welcome, since I don't have an IMU for testing at the moment. :)

thejeshk, 2019-01-31:
Hi Mintar, 

I want to integrate the "4x Isolated Digital Input Phidget" to :::ROS 

are there any drivers or a package for that particular phidget? 

Sorry if my question is silly. i am very new to :::ROS

mintar, 2019-01-31:
@thejeshk Your comment has nothing to do with this issue, so I've created a new issue for it: #31 . Please continue the discussion there.

mintar, 2022-02-16:
I'm going to close this because diagnostics support was removed in Noetic (see #43). If somebody ever gets around to re-adding diagnostics support, we should make sure that we resolve this issue as well.

closed 2022-02-16 by mintar