ros-drivers/phidgets_drivers #31: Add support for "4x Isolated Digital Input Phidget"

mintar, 2019-01-31:
Hi Mintar, 

I want to integrate the "4x Isolated Digital Input Phidget" to :::ROS 

are there any drivers or a package for that particular phidget? 

Sorry if my question is silly. i am very new to :::ROS

_Originally posted by @thejeshk in https://github.com/ros-drivers/phidgets_drivers/issues/29#issuecomment-459280055_

mintar, 2019-01-31:
Hi @thejeshk, I don't know if there are any ROS packages for that device yet. If you want to add support for it yourself, this is what you would have to do:

* Add a new class to phidgets_api that uses the libphidgets to talk to your device. Example:

  - [phidgets_api/include/phidgets_api/encoder.h](https://github.com/ros-drivers/phidgets_drivers/blob/782f6bde2334cf1ad8e709c2022f2f21a2935a0e/phidgets_api/include/phidgets_api/encoder.h)
  - [phidgets_api/src/encoder.cpp](https://github.com/ros-drivers/phidgets_drivers/blob/782f6bde2334cf1ad8e709c2022f2f21a2935a0e/phidgets_api/src/encoder.cpp)

* Add a new ROS package that calls the functions you've just created in phidgets_api. Example:

  - [phidgets_high_speed_encoder](https://github.com/ros-drivers/phidgets_drivers/tree/782f6bde2334cf1ad8e709c2022f2f21a2935a0e/phidgets_high_speed_encoder)

thejeshk, 2019-02-18:
Hi @mintar 
Firstly Sorry for late reply. I had examinations and I paused doing this project until last week.  
I am very much new to Programming my background is Mechanical Engineering. Though I started learning Python and C++ I am not getting exactly what to do. 

Would you please elaborate how I could create a class for particularly this "4x Isolated Digital Input Phidget" 

I am sorry if I sound silly. 

Thank you 
TK

mintar, 2019-08-15:
This will probably be fixed by #39 (in ROS noetic).

closed 2019-08-15 by mintar