ros-drivers/phidgets_drivers #32: Servo question

zhughesjd, 2019-02-09:
So this is more of a question than an issue.
I bought the PhidgetAdvancedServo 8-Motor controller hoping to use it from ROS as described here:

http://wiki.ros.org/phidgets

From there was then directed here to use this latest phidgets_driver repo.
I cannot an example launch file or an example rosrun command to use this controller.
Can someone direct me to one of these examples?
BTW I am using ROS Kinetic.

Thanks
Joshua

mintar, 2019-02-11:
The PhidgetAdvancedServo device is not yet supported by phidgets_drivers. If you own this device and would like to add it to phidgets_drivers, I'd be happy to review a pull request.

zhughesjd, 2019-02-11:
Ah... Got it. Thanks mintar. If my project comes to fruition, will let you know

closed 2019-02-11 by zhughesjd