ros-drivers/phidgets_drivers #35: Switch to libphidget22

clalancette, 2019-07-15:
I'm interested in adding support for some additional Phidgets device, namely the [VINT hub](https://www.phidgets.com/?tier=3&catid=2&pcid=1&prodid=643) (and some sensors behind it).  Unfortunately, the VINT hub is not supported by libphidget21, but it is supported by libphidget22.

I've done some preliminary work to see how hard it would be to switch the current code to libphidget22.  The major issue as I see it is that libphidget22 has a very different idea of how to connect to devices.  For instance, instead of having an `InterfaceKit` structure, and having the Digital Inputs and Outputs attached to that, there seems to instead be a separate `DigitalInput` and `DigitalOutput` structure.  This is more flexible, but also means that the current class structure in `phidgets_api` doesn't really match.

Assuming we want to move to libphidget22, I can see two different ways we can make this work:
1.  Keep the current class structure, "emulating" the old ways of doing things by creating multiple device handles inside the IK class, for instance.  This has the advantage of looking like the current structure, but trades off a bunch of flexibility.
1.  Change the class structure to more closely match how libphidget22 works.  This has the advantage of flexibility, but it means that all of the old APIs, nodes, launch files, etc. would change.

Ultimately I want to port this entire thing to ROS 2, so backwards compatibility isn't a huge concern to me.  Thus I'd personally choose option 2, as it gives us access to larger numbers of devices.  However, I'd like to get some input/advice before moving forward, since I'd like to submit whatever I do to this repository.  Thanks in advance for your thoughts.

mintar, 2019-07-16:
Hi Chris, this sounds like a great idea! I'm fine with option 2; it probably means less technical debt. Also option 1 means we would need to invent APIs for the devices not supported by libphidget21, so we would end up with a horrible hybrid.

I suggest you just go with option 2 without regard for backwards compatibility at first, and then we can add node shims that emulate the old interface, only in order not to break compatibility for already released versions. In a future ROS version (Noetic, ROS 2) we can drop the shims. Sounds like a plan?

BTW, I'm on vacation until August 5, so I might be a bit slow to respond.

clalancette, 2019-07-17:
Thanks for the feedback, it's appreciated.  I have some updates to the current code that I'll open PRs for, then I'll start work on the new classes using libphidget22.

mintar, 2019-07-19:
That's great! Open those PRs whenever they are ready, I'll work on them when I am back.

clalancette, 2019-09-05:
Closing this, as it is now merged :fireworks:

closed 2019-09-05 by clalancette