ros-drivers/phidgets_drivers #40: IMU gyro calibration time is not taken into account in the ROS service.

MichaelGrupp, 2019-08-13:
The `CPhidgetSpatial_zeroGyro` function from the Phidgets API returns directly, however the actual process takes around 2 seconds as written in the [documentation](https://www.phidgets.com/documentation/web/cdoc/group__phidspatial.html#ga066d93dd4a7d52f439e728e7cc64ca3):

> Zeroes the gyroscope. This takes about two seconds and the gyro zxes will report 0 during the process. This should only be called when the board is not moving. 

The `imu/calibrate` service logs success and returns immediately after calling this gyro calibration. This is bad because to a client it looks like the calibration is done, although moving the IMU in the 2 seconds period right after the calibration call will lead to a constant drift.

https://github.com/ros-drivers/phidgets_drivers/blob/37b79e32381adc1015cafcbff65d52472a2caac9/phidgets_imu/src/imu_ros_i.cpp#L212-L214

---

I propose the following:

1. change the log messages of the ROS service

2. change the behavior of the service handler?
Block 2 seconds or introduce an API change and use an action that takes 2 seconds?

Side note: Currently the service returns an empty response but there's an extra, unlatched `/imu/is_calibrated` topic that kind of takes over the job of a service response field. If we change the service API, we could also get rid of that topic by introducing a proper response field. Or, make it latched so one can check when it was calibrated the last time?

mintar, 2019-08-14:
Let's continue the discussion in #41 so it's all in one place.

mintar, 2019-08-15:
Fixed via #41 .

closed 2019-08-15 by mintar