ros-drivers/phidgets_drivers #62: Bloom-Release to ROS2 Dashing / Eloquent

IanTheEngineer, 2019-10-23:
Thank you for all the re-working done in this repo for ROS2 @clalancette & @mintar! I've been building this repo from source and using it with several sensors for a few weeks now. So far in my testing, things seem to be working well, though I admittedly am not using every phidget sensor. 

This is just a friendly request to push these packages up into the ros buildfarm for ease of use/deployment:

- [ ] `bloom-release` into `ros-dashing`
- [ ] `bloom-release` into `ros-eloquent`

Thanks!

clalancette, 2019-12-03:
@mintar I'm preparing to do this release.  I've created https://github.com/ros2-gbp/phidgets_drivers-release to get it ready.  One question for you; what version number would you like to see on the ROS 2 port?  The version released into Melodic is 0.7.9; for ROS 2 ports, we've mostly been bumping the major number to make sure there is never a tag overlap between ROS 1 and ROS 2 releases.  However, there is a reasonable argument to make that we should bump the major number for Noetic (since it is a large rewrite).

Thus, I'm proposing that the version strategy for this repository is as follows:
- 0.x.y - Melodic and earlier releases
- 1.x.y - Noetic release
- 2.x.y - ROS 2 releases (Dashing and Eloquent for now).

What do you think of this proposal?

mintar, 2019-12-05:
I fully agree with your proposal of the version strategy, let's do it this way.

Just out of curiosity: The ROS1 release repository is at https://github.com/ros-drivers-gbp/phidgets_drivers-release. What's the best practice for ROS2 release repos: Reusing the ROS1 release repos or creating a new one?

clalancette, 2019-12-05:
> I fully agree with your proposal of the version strategy, let's do it this way.

Great, thanks for the feedback.  I'll proceed this way.

> What's the best practice for ROS2 release repos: Reusing the ROS1 release repos or creating a new one?

As in ROS 1, in ROS 2 it is possible to host the release repository anywhere (though we highly recommend not using the source repository as the release repository as well).

But for the most part, what we've been doing for ROS 2 is to create an entirely new release repository under the https://github.com/ros2-gbp organization.  This keeps the release meta-information separated between the ROS 1 and ROS 2 ports.  That's what I've chosen to do here.

clalancette, 2019-12-05:
Released 2.0.0 for ROS 2 stuff, and PRs opened:

* Dashing - https://github.com/ros/rosdistro/pull/23227
* Eloquent - https://github.com/ros/rosdistro/pull/23228

clalancette, 2019-12-05:
And I've now merged both of those.  I'm going to close this out in anticipation that those builds complete successfully, and the packages will now be available in the testing repository.  If that turns out not to be the case, I'll re-open this.

closed 2019-12-05 by clalancette