ros-drivers/phidgets_drivers #69: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container

Chiasung, 2020-03-12:
Hello All,

Recently, I switch my project from ROS1 to ROS2.
After I ran 
`ros2 launch  phidgets_spatial spatial-launch.py`.

I got the following error messages. Then the process is stuck there.
Any help to solve this issue, I will greatly appreciate. 

OS: Ubuntu 18.04 in docker container
branch: dashing

- [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container '/phidget_container': Component constructor threw an exception
- [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '' in container '/phidget_container'
- [component_container-1] [INFO] [phidget_container]: Load Library: /module/install/phidgets_spatial/lib/libphidgets_spatial.so
- [component_container-1] [INFO] [phidget_container]: Found class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
- [component_container-1] [INFO] [phidget_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
- [component_container-1] [INFO] [phidgets_spatial]: Starting Phidgets Spatial
- [component_container-1] [ERROR] [phidget_container]: Component constructor threw an exception
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
- [component_container-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
- [INFO] [component_container-1]: process has finished cleanly [pid 10351]

clalancette, 2020-03-12:
So you are getting to the constructor of the class:

```
[component_container-1] [INFO] [phidgets_spatial]: Starting Phidgets Spatial
```

But you are not getting to where it tries to connect to the device.  The next line you should see is something like:

```
[component_container-1] [INFO] [phidgets_spatial]: Connecting to Phidgets Spatial serial -1, hub port 0 ...
```

The only things that happen between those two prints are getting and checking of parameters.  So I'd guess that one of your parameters is either declared as an invalid type, or with an invalid value.  What does your parameter configuration file look like?

Chiasung, 2020-03-13:
Hi @clalancette ,

 Here is my params.yaml and launch file.
```yaml
phidgets_spatial:
  ros__parameters:
    hub_port: 0
    frame_id: "imu_link"
    linear_acceleration_stdev: 0.002744
    angular_velocity_stdev: 0.087266463
    magnetic_field_stdev: 0.0000011
    time_resynchronization_interval_ms: 5000
    data_interval_ms: 8
    callback_delta_epsilon_ms: 1
    publish_rate: 100
```

```python
# Copyright 2019 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch a Phidgets spatial in a component container."""

import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

"""Load Configuration File"""
param_file = os.getcwd() + '/params.yaml'
node_remaps=[]
if os.path.isfile(param_file):
    node_remaps += [('__params', param_file)]
    launch_print("Using parameters file: " + param_file)
else:
    launch_print("Parameters file not found: " + param_file)

def generate_launch_description():
    """Generate launch description with multiple components."""
    container = ComposableNodeContainer(
            node_name='phidget_container',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='phidgets_spatial',
                    node_plugin='phidgets::SpatialRosI',
                    node_name='phidgets_spatial'),
            ],
            output='both',
    )

    return launch.LaunchDescription([container])

```

clalancette, 2020-03-13:
```
publish_rate: 100
```

That's your problem.  Currently ROS 2 doesn't accept integers for double fields, and the publish rate is a double.  Change that to `100.0`, and it will most likely work.

I'm going to close this out, but feel free to reopen if this doesn't fix your problem.

closed 2020-03-13 by clalancette