ros-drivers/phidgets_drivers #71: compass orientation

mturktest123, 2020-05-19:
Can someone provide a link to the ROS code to get compass orientation from the phidgets spatial (IMU)? I can find the raw magnetometer data (xyz) code but am looking for the compass direction.

https://www.phidgets.com/?tier=3&catid=10&pcid=8&prodid=1025

Thanks, Allan

clalancette, 2020-05-19:
> Can someone provide a link to the ROS code to get compass orientation from the phidgets spatial (IMU)? I can find the raw magnetometer data (xyz) code but am looking for the compass direction.

Unless I'm misreading the documentation, the libphidgets API doesn't provide this information (and hence, the ROS wrapper here doesn't either).  It only provides for the raw magnetometer readings.  You could combine the raw magnetometer data with the acceleration data to do it, but I understand it is pretty tricky.  If you need this information, an easier solution may be to get an IMU that does these calculations onboard.

mturktest123, 2020-05-19:
phidgets provides a partial piece of code in c that would provide compass direction; but it would have to be rewritten to make it work in ROS - so I was trying to avoid doing that...  https://www.phidgets.com/docs/Magnetometer_Primer#Calculating_Heading_from_Magnetometer_Data

clalancette, 2020-05-19:
> phidgets provides a partial piece of code in c that would provide compass direction; but it would have to be rewritten to make it work in ROS - so I was trying to avoid doing that... https://www.phidgets.com/docs/Magnetometer_Primer#Calculating_Heading_from_Magnetometer_Data
> (code is at the bottom of the page this link takes you to)

Ah, I see.  Yeah, you'll have to write that code to take in the data from ROS and calculate it yourself.  There might be *some* argument for including that as part of the phidgets_spatial package, but I'm not entirely sure if we should do that.  @mintar thoughts on that?

mturktest123, 2020-05-19:
thanks for checking into this. perhaps an additional (better?) question for me to ask is 'is there a link to a recommended ROS node that should use to bring the phidget magnetometer data into move_base or move_base_flex'?

mintar, 2020-05-19:
Look at this launch file: https://github.com/ros-drivers/phidgets_drivers/blob/melodic/phidgets_imu/launch/imu.launch

That launch file passes the messages from the phidgets driver to the imu_filter_madgwick, which fills out the orientation field in the message. You should pass that message (along with the wheel odometry) to robot_localization, which will fuse the information and publish a odom -> base_link tf transform. Then you'll probably also start amcl localization, which publishes the map -> odom transform. Together they give you the map -> base_link transform, which will be used by move_base or move_base_flex.

The magnetometer is pretty much useless on a wheeled indoor robot though, so I would recommend disabling it and only use the gyro and accelerometer.

mturktest123, 2020-05-19:
thanks for the great overview! and the links! I'll close this issue and follow up later in a new issue if I have more questions.

closed 2020-05-19 by mturktest123