ros-drivers/phidgets_drivers #81: set_digital_output service has success=false, but output is set correctly

danneboom, 2020-12-04:
I have a PhidgetInterfaceKit 8/8/8 P/N1018. When I run

````
~$ rosrun phidgets_ik  phidgets_ik_node 
[ INFO] [1607075109.917870379]: Starting Phidgets IK
[ INFO] [1607075109.918402573]: Opening device
[ INFO] [1607075109.918626262]: Waiting for IK -1 to be attached...
Phidget attached (serial# 324112)
[ INFO] [1607075109.938885559]: 8 inputs, 8 outputs, 8 sensors
[ INFO] [1607075125.475819619]: Setting output 0 to 1
````

and then in another terminal
````
~$ rosservice call /set_digital_output "index: 0 
state: true" 
success: False
````

the success variable is false, but the output is actually set correctly. Why "success" is false?

clalancette, 2020-12-04:
From this report, I'm assuming that you are using phidgets on Kinetic or Melodic.  Can you please confirm which one you are using?

Assuming that is true, than the code that does the setting is here: https://github.com/ros-drivers/phidgets_drivers/blob/f94866caecefda08a6f22b407d0c04b24a054db7/phidgets_ik/src/ik_ros_i.cpp#L96

You can see that it sets `res.success` to the inverse of what the state is.  It is kind of odd behavior, I agree.  Maybe @mintar can shed some light on why it was done that way originally?

I will point out that in the noetic branch (and the in the ROS 2 branches), the state gets set properly and success always gets set to `true`.

mintar, 2020-12-04:
That bug seems to have been introduced in 21109218:

* A new function `IK::setOutputState` was introduced which returns `!CPhidgetInterfaceKit_setOutputState` (notice the "**!**"):

https://github.com/ros-drivers/phidgets_drivers/commit/211092188e7b829e10614abd7e261de353091648#diff-dd8b044942a7eb33f665646f51775f09d54e8cf6880d53a05b70f508152d55d4R55-R59

* The function call to `!CPhidgetInterfaceKit_setOutputState` was replaced by `!setOutputState`, but it should have been just `setOutputState`:

https://github.com/ros-drivers/phidgets_drivers/commit/211092188e7b829e10614abd7e261de353091648#diff-7e6292f6c639563aabec4ae31525423eb0f72156f927be1a013f4487eafbc3a7L96-R95

I've just pushed a commit that fixes this (ca16fca). I'm closing this issue, but feel free to reopen if you want to continue the discussion.

closed 2020-12-04 by mintar

clalancette, 2020-12-04:
> [](https://github.com/ros-drivers/phidgets_drivers/commit/211092188e7b829e10614abd7e261de353091648#diff-7e6292f6c639563aabec4ae31525423eb0f72156f927be1a013f4487eafbc3a7L96-R95)
>
> I've just pushed a [...]

Thanks!

commit ca16fca8 (2020-12-04): Fix success flag of phidgets_ik set_digital_output service

Fixes #81