ros-drivers/phidgets_drivers #83: Bridge 4 Input Support

ChristianMarzi, 2021-03-07:
Hi 
this is a copy of my post on [ROS answers](https://answers.ros.org/question/373291/phidgetbridge-4-input-support/), but i think here might be a better place to ask: 

im trying to get the data from the [PhidgetBridge 4-Input](https://www.phidgets.com/?tier=3&catid=98&pcid=78&prodid=1027) into ros noetic (e.g. publishing the sensor values on a topic). As far as i've understood this device is not supported yet. Is that correct or could i reuse one of the existing packes in [phidgets_drivers](https://github.com/ros-drivers/phidgets_drivers/tree/noetic)? Maybe with minor modification?

So far i've tried using the analog input nodelet `roslaunch phidgets_analog_inputs analog_inputs.launch`(pulled and built the noetic branch) however i got the connection problem: `[ERROR] [1614854548.863487183]: AnalogInputs: Failed to open device: Timed Out`(device is pluged in usb port an visible in /dev)

If unsupported, i'm willing to try to contribute, however as i've never done anything alike i would appriciate any reccomendation or help on what steps to take?

mintar, 2021-03-08:
The `phidgets_analog_inputs` package was written with the [PhidgetInterfaceKit 8/8/8](https://www.phidgets.com/?tier=3&catid=2&pcid=1&prodid=1021) or [PhidgetInterfaceKit 2/2/2](https://www.phidgets.com/?tier=3&catid=2&pcid=1&prodid=4) in mind, but I believe you should be able to make it work for the PhidgetBridge as well with some minimal changes.

I believe the main change is here:

https://github.com/ros-drivers/phidgets_drivers/blob/8f3733462e65840231f6ae53b3791b250d9fbb9d/phidgets_api/src/analog_inputs.cpp#L65-L66

In this line, we're trying to open a device with ChannelClass `PHIDCHCLASS_VOLTAGEINPUT`. According to the API, the PhidgetBridge has a ChannelClass of `PHIDCHCLASS_VOLTAGERATIOINPUT` instead. This is because the Bridge doesn't provide analog inputs, but instead voltage ratio inputs.

It's probably best if you look at the following files:

```
phidgets_api/include/phidgets_api/analog_input.h
phidgets_api/include/phidgets_api/analog_inputs.h
phidgets_api/src/analog_input.cpp
phidgets_api/src/analog_inputs.cpp
phidgets_analog_inputs/*
```

Then copy all `*analog_input*` to something new, like `*voltage_ratio_input*` and make the relevant changes there.

ChristianMarzi, 2021-03-09:
Thanks for the input. I now have a working version locally. I'll clean it up and create a pull request.