ros-drivers/phidgets_drivers #87: Phidgets Imu driver splits my tf tree

Oburshan, 2021-03-25:
Hey, I am using phidgets imu 3/3/3 on google cartographer, and when i launch the imu driver it splits my tf tree which avoid cartographer node to provide a map because he don't have a complete tf tree.
The launch: `roslaunch phidgets_imu imu.launch`

Before Launching imu driver (Only cartographer node executed):
[image]

And after:
[image]

Thanks!

Oburshan, 2021-03-25:
I solved it after removing the ImuFilterNodelet node from my launch file, i will close it now.

closed 2021-03-25 by Oburshan

mintar, 2021-03-25:
If you don't need the orientation, you can just remove the nodelet, as you just did. Alternatively, the ImuFilterNodelet has a parameter `publish_tf` that you should set to `false`.