ros-drivers/phidgets_drivers #92: Add usage instructions to README

tim-fan, 2021-05-26:
Hi there, I'm just trying this out in ROS2 for the first time with an IMU. All is working well, thank you to all the developers!

Just thought I'd note I  had a bit of difficulty figuring out how to get the driver running, and maybe an addition to the readme would help others.

Basically I expected to be able to run a `phidgets_spatial` node, and I was a bit confused when I found the package does not build any nodes (only components). Later I found that there are launch files which handle the launching of the component.

My suggestion would be to add to the `phidgets_spatial` `README.md`:
```
## Usage
To launch the phidgets_spatial node:
ros2 launch phidgets_spatial spatial-launch.py
```
I assume a similar addition could be made to the rest of the driver packages as well.

---

Extra detail: The sequence of steps I took to get it going:

1. Look at the phidget spatial readme, which documents the node API, but not how to start the node
2. try to `ros2 run phidget_spatial phidget_spatial`, but that executable does not exist
3. try to figure out which executables exist in the phidget spatial package, but there are none: `ros2 pkg executables phidgets_spatial`
4. Have a look into the CMakeLists.txt, notice that no nodes are built, but a component is registered
5. Google ros2 components, found this https://ubuntu.com/blog/components-vs-plugins-in-ros-2
6. Based on the above, started a component container and load the phidgets_spatial component
7. success!
8. notice from reading issue logs that phidgets_spatial that launch files exist for starting the node
9. more convenient success!

mintar, 2021-05-26:
Thank you for your very nice report and the extra details, this kind of feedback from users is very helpful! When you've worked on a project for a long time, it's sometimes not easy to see which parts of the documentation new users struggle with.

@clalancette : Since this concerns ROS2, could you have a look at this?

clalancette, 2021-05-26:
@tim-fan Thanks for the detailed feedback.  I agree that the documentation could be better here.

It turns out that with "modern" phidgets, we cannot have individual binaries.  The reason for this is that when you use a phidget VINT hub, you will have multiple devices connected to the hub, but only one USB connection from that hub to the host computer.  When that is the case, you can only have *one* process that accesses the USB connection at a time.  So in order to access all of the devices behind the hub, the ROS 2 phidgets driver is designed to load multiple components into one process.  We decided not to provide individual binaries to simplify the story.

But we obviously don't have enough information about this in the README.  I'm going to do a small pass over the top-level README at https://github.com/ros-drivers/phidgets_drivers/blob/galactic/README.md and add some additional information.

PR #93 (2021-05-27): Update the ROS 2 readme files.
This adds more information about running the nodes.

@tim-fan I'd appreciate a look over this by you to see if it gives you the information you needed.

Fixes #92