ros-drivers/phidgets_drivers #94: [galactic] spatial-launch.py fails when not providing Magnetometer corrections

karl-schulz, 2021-07-12:
## Bug Report
After the switch to ROS2 Galactic, the `spatial-launch.py` (and probably also the magnetometer) fails when not providing magnetometer corrections.

### Reproduction
Simply calling `ros2 launch phidgets_spatial spatial-launch.py` yields:
```
[INFO] [component_container-1]: process started with pid [4695]
[component_container-1] 1626088543.095012 [0] component_: using network interface eth0 (udp/192.168.1.210) selected arbitrarily from: eth0, docker0
[component_container-1] [INFO] [1626088543.296766218] [phidget_container]: Load Library: /home/angsa/ros_ws/install/phidgets_spatial/lib/libphidgets_spatial.so
[component_container-1] [INFO] [1626088543.309701892] [phidget_container]: Found class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1626088543.310028374] [phidget_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1626088543.336890208] [phidgets_spatial]: Starting Phidgets Spatial
[component_container-1] [ERROR] [1626088543.536905484] [phidget_container]: Component constructor threw an exception: expected [double] got [not set]
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container '/phidget_container': Component constructor threw an exception: expected [double] got [not set]
```

### Reason
The declare_parameter here fails:
https://github.com/ros-drivers/phidgets_drivers/blob/0a4e583e9d60c5c748e098837d11afafb8171ce5/phidgets_spatial/src/spatial_ros_i.cpp#L110

The error is then thrown here: 
https://github.com/ros2/rclcpp/blob/d0fa844a093b26e34dc0989e1b3ab94aee5ba896/rclcpp/include/rclcpp/parameter_value.hpp#L66

I propose moving the `declare_parameter` in the `try..catch` clause and additionally catch the `rclcpp::ParameterTypeException`.

We did that in https://github.com/angsa-robotics/phidgets_drivers

clalancette, 2021-07-12:
@karl-schulz See #95, which I believe will fix the issue.  Any testing that you can do for that is appreciated!

karl-schulz, 2021-07-12:
Tested it and it fixes it! Thanks!

PR #95 (2021-08-03): Make the magnetometer corrections optional again.
Previously, I had made the code do the following:

this->declare_parameter<double>("cc_mag_field")

which gets rid of a deprecation warning on Galactic.  Unfortunately,
this also makes the parameter required.  Instead, we can do
the following, which keeps the parameter optional but also
gets rid of the deprecation warning:

this->declare_parameter("cc_mag_field", rclcpp::PARAMETER_DOUBLE)

This should fix #94