ros-drivers/phidgets_drivers #96: Failed to load library (ros foxy installation)

Edwin-Til-Illan, 2021-08-04:
After installing phidgets_drivers from source I tried to launch the spatial executable using:
`ros2 launch phidgets_spatial spatial-launch.py`
And I got:
```
[INFO] [launch]: All log files can be found below /home/rover/.ros/log/2021-08-04-15-35-36-796098-rover-ThinkCentre-M73-39388
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [39401]
[component_container-1] [INFO] [1628105737.177545698] [phidget_container]: Load Library: /home/rover/rover_ws/install/phidgets_spatial/lib/libphidgets_spatial.so
[component_container-1] [ERROR] [1628105737.180257760] [phidget_container]: Failed to load library: Could not load library LoadLibrary error: libphidget22.so.0: cannot open shared object file: No such file or directory, at /home/rover/ros2_foxy/src/ros2/rcutils/src/shared_library.c:84
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container '/phidget_container': Failed to load library: Could not load library LoadLibrary error: libphidget22.so.0: cannot open shared object file: No such file or directory, at /home/rover/ros2_foxy/src/ros2/rcutils/src/shared_library.c:84
```
I checked and there is a 'shared_library.c' at path '/home/rover/ros2_foxy/src/ros2/rcutils/src/'
Any clue what is going wrong ?

clalancette, 2021-08-06:
@Edwin-Til-Illan Can you give https://github.com/ros-drivers/phidgets_drivers/pull/97 a shot and see if it solves the problem for you?

Edwin-Til-Illan, 2021-08-06:
Thanks for the help !
However I just did not install libphidget22 on my OS, problem solved:
[https://www.phidgets.com/docs/OS_-_Linux](url)

closed 2021-08-06 by Edwin-Til-Illan

clalancette, 2021-08-08:
For what it is worth, you should not have to install `libphidget22` on your own; this repository should do it for you.  That's what #97 is fixing.  But glad you got it working otherwise.