ros-drivers/phidgets_drivers #98: PhidgetDCMotor_getBackEMF Operation Not Supported Noetic

PatrickAirbus, 2021-08-15:
i get this problem when i try to launch the motor lauch file.
I run ubuntu in Virtuell Box. Motor Phidget DCM4002_0 with the controller DCC1000_0 and the Hub HUB0000_0.

clalancette, 2021-08-16:
I'm sorry, you'll need to provide a lot more information about what trouble you are seeing.  What Ubuntu version are you using?  What ROS version are you using?  What are the commands you ran?  What is the output you are getting?

Patrick-Neo, 2021-08-17:
I´m using Ubuntu 20.04.2.0 ROS Noetic Ninjemys. I ran the motor.launch file the error was this 

"[ERROR] [1628689658.891669262]: Motors: Failed to get back EMF for Motor channel 0: Operation Not Supported [PhidgetsMotorsNodelet-3] process has died [pid 6245, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load phidgets_motors/PhidgetsMotorsNodelet motors_manager __name:=PhidgetsMotorsNodelet __log:=/home/patrick/.ros/log/ae703c8a-faaa-11eb-ba7c-ed12fa709ec8/PhidgetsMotorsNodelet-3.log]."

i trackt the problem down in the file motors_ros_i.cpp line 114
"motor_vals_[i].last_back_emf_val = motors_->getBackEMF(i);"

any more infos needed?

mintar, 2022-02-11:
Closing due to inactivity. If your problem has not been solved in the mean time, feel free to re-open this issue!

closed 2022-02-11 by mintar