ros-drivers/phidgets_drivers #105: Failed to load nodelet '/PhidgetsSpatialNodelet` of type `phidgets_spatial/PhidgetsSpatialNodelet` to manager `imu_manager

DoneMan, 2021-10-20:
did anyone encounter this before?

i have the phidgets spatial 3/3/3 MOT0109_0, i'm trying to launch the spatial.launch and i get these errors:
Spatial: Failed to open device: Timed Out
[FATAL] [1634711799.897689893]: Failed to load nodelet '/PhidgetsSpatialNodelet` of type `phidgets_spatial/PhidgetsSpatialNodelet` to manager `imu_manager'
[ INFO] [1634711799.907888615]: Initializing IMU Filter Nodelet
[ INFO] [1634711799.910292674]: Starting ImuFilter
[ INFO] [1634711799.921251294]: Using dt computed from message headers
[ INFO] [1634711799.921352759]: The gravity vector is kept in the IMU message.
[ INFO] [1634711799.941848213]: Imu filter gain set to 0.100000
[ INFO] [1634711799.941937589]: Gyro drift bias set to 0.000000
[ INFO] [1634711799.941980552]: Magnetometer bias values: 0.000000 0.000000 0.000000
[PhidgetsSpatialNodelet-3] process has died [pid 376, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load phidgets_spatial/PhidgetsSpatialNodelet imu_manager __name:=PhidgetsSpatialNodelet __log:=/home/user/.ros/log/13e95abe-3170-11ec-92fd-b05cdaa8ff51/PhidgetsSpatialNodelet-3.log].
log file: /home/user/.ros/log/13e95abe-3170-11ec-92fd-b05cdaa8ff51/PhidgetsSpatialNodelet-3*.log

can anyone help please?

phidgets, 2021-10-20:
MOT0109_0 requires phidget22. The module will need to be migrated from phidget21 to phidget22.

DoneMan, 2021-10-21:
@mintar it works!!!  you are a freaking king! thank you very much!!

closed 2021-10-21 by DoneMan