ros-drivers/phidgets_drivers #109: Voltage Inputs of VINT Hub

leon-no37, 2021-12-08:
May I check with you on how to use the VINT hub to get the voltage inputs with ROS?

Thanks!

clalancette, 2021-12-08:
Generally, you would instantiate a `phidgets_digital_input`, similar to what is done in https://github.com/ros-drivers/phidgets_drivers/blob/noetic/phidgets_digital_inputs/launch/digital_inputs.launch (for ROS 1) or https://github.com/ros-drivers/phidgets_drivers/blob/galactic/phidgets_digital_inputs/launch/digital_inputs-launch.py (for ROS 2).  You can then get the state of the digital input by reading the `digital_input0` (or `digital_input1`, etc), which corresponds to the input line.

leon-no37, 2021-12-09:
I have received this error message "DigitalInputs: Failed to open device: Timed Out", when I try to run "roslaunch phidgets_digital_inputs digital_inputs.launch". May I know how to solve this problem?

> Generally, you would instantiate a `phidgets_digital_input`, similar to what is done in [...]

clalancette, 2021-12-09:
It's hard to say what the problem is.  Did you setup the udev rules as specified in https://github.com/ros-drivers/phidgets_drivers#udev-rules-setup ?

leon-no37, 2021-12-09:
> It's hard to say what the problem is. Did you setup the udev rules as specified in https://github.com/ros-drivers/phidgets_drivers#udev-rules-setup ?

Ok, thanks!

closed 2021-12-09 by leon-no37