ros-drivers/phidgets_drivers #110: Cannot specify remote vint hubs

J-C-Haley, 2022-01-05:
The drivers do not discover a vint hub without specifying server details, i.e.:

PhidgetNet_addServer("myServerVintHub","100.100.0.1", 5661, "", 0);

Inserting that line and adding rosparams for server_name and server_ip, I have it working on the phidgets_temperature package. If it's agreeable to the devs, I can add similar launch file parameters to all the packages and submit a PR?

In addition, a similar change I could include would be to specify a server_password parameter from the launch file and the 'setServerPassword' method. 

Please let me know if this is useful and I'll push, thank you!

clalancette, 2022-01-06:
Seems like a reasonable feature to add, so :+1: from me.

mintar, 2022-02-11:
Yes, sounds like a welcome feature! Looking forward to your pull request!

J-C-Haley, 2022-04-05:
Took me a while, but I got around to adding this in & copypasting to all the devices, see the PR!

closed 2022-04-05 by J-C-Haley