ros-drivers/phidgets_drivers #111: PhidgetSpatial 3/3/3 Basic 1042_0B Connection Issues - Melodic

austin-InDro, 2022-01-20:
Hello,

I am having trouble using your phidget_imu imu.launch script.
Issue:
`roslaunch phidget_imu imu.launch`
Outputs the following

SUMMARY
========

PARAMETERS
 * /ImuFilterNodelet/use_mag: True
 * /ImuFilterNodelet/use_magnetic_field_msg: True
 * /PhidgetsImuNodelet/period: 4
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    ImuFilterNodelet (nodelet/nodelet)
    PhidgetsImuNodelet (nodelet/nodelet)
    imu_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[imu_manager-1]: started with pid [9468]
process[PhidgetsImuNodelet-2]: started with pid [9469]
process[ImuFilterNodelet-3]: started with pid [9470]
[ INFO] [1642688272.830204431]: Loading nodelet /PhidgetsImuNodelet of type phidgets_imu/PhidgetsImuNodelet to manager imu_manager with the following remappings:
[ INFO] [1642688272.865155316]: Loading nodelet /ImuFilterNodelet of type imu_filter_madgwick/ImuFilterNodelet to manager imu_manager with the following remappings:
[ INFO] [1642688272.881463024]: Initializing nodelet with 4 worker threads.
    }
[ INFO] [1642688272.886425733]: waitForService: Service [/imu_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1642688272.913038285]: waitForService: Service [/imu_manager/load_nodelet] is now available.
[ INFO] [1642688272.963427972]: Initializing Phidgets IMU Nodelet
[ INFO] [1642688272.994070524]: Starting Phidgets IMU
[ INFO] [1642688273.089416149]: Opening device
[ INFO] [1642688273.089530837]: Waiting for IMU to be attached...
**[FATAL] [1642688283.093874271]: Problem waiting for IMU attachment: Given timeout has been exceeded. Make sure the USB cable is connected and you have executed the phidgets_api/share/setup-udev.sh script.**
[ INFO] [1642688283.094001406]: Calibrating IMU, this takes around 2 seconds to finish. Make sure that the device is not moved during this time.
[ INFO] [1642688285.094174895]: Calibrating IMU done.

Additional Info:
- I can interact with the device using the Phidget Python library, so I know the device is connected and working.
- I've installed the libphidget22 and libphidget21 libraries separately as well to ensure that when the system is compiled it would have everything it needs. However, i've noticed the ros package comes with it's own libphidget21 folder (so I assume the system libraries aren't used)
Device:
- Jetson Nano
OS:
- Ubuntu 18.04

Any help would be greatly appricated.
Thanks,

mintar, 2022-01-21:
You can try building the `noetic` branch of this repo from source. It also compiles under melodic and includes a newer version of the libphidget22 library.

austin-InDro, 2022-01-21:
> You can try building the `noetic` branch of this repo from source. It also compiles under melodic and includes a newer version of the libphidget22 library.

Hi Mintar, sorry forgot to mention that I original used the noetic branch (worked fine on my machine running noetic). However, it didn't seem to pick up the Phidget on the Jetson. I noticed this comment on the PhidgetSpatial site for the MOT0109_0  version,

`Due to parts and components going obsolete and becoming unavailable, the MOT0109_0 uses a different processor than the 1044_1B that is not supported under Phidget21. If you're using our old libraries you must upgrade to Phidget22 before you can make use of this Phidget.`

I believe this was the issue for using the Melodic branch for the MOT0109_0 specifically. Either way, I've created my own node using python so all is good now.

Thanks.

austin-InDro, 2022-01-28:
I went back to this library and tried your suggestion Martin. It seems third time is the charm. Now everything is working on the Noetic branch...

Thanks.

closed 2022-01-28 by austin-InDro