ros-drivers/phidgets_drivers #122: Allow configuring encoder IOMode

lights0123, 2022-03-17:
The new encoder module ENC1000_0 has several [IOMode options](https://www.phidgets.com/docs/Encoder_Primer#Output_Circuit):

```cpp
typedef enum {
 ENCODER_IO_MODE_PUSH_PULL = 0x1,
 ENCODER_IO_MODE_LINE_DRIVER_2K2 = 0x2,
 ENCODER_IO_MODE_LINE_DRIVER_10K = 0x3,
 ENCODER_IO_MODE_OPEN_COLLECTOR_2K2 = 0x4,
 ENCODER_IO_MODE_OPEN_COLLECTOR_10K = 0x5,
} Phidget_EncoderIOMode;
```

This should be configurable from the node. I'd like to add it as a node parameter (and I'll send a PR), but I'd like to know should this be configured as a string (and match it to the enum) or an integer and just pass it directly?

clalancette, 2022-03-22:
I'd personally go for a string; that at least has semantic meaning and is easier for the user to use.  @mintar thoughts?

mintar, 2022-04-08:
Sorry, I totally forgot about this one. I agree, I'd go with a string too.