ros-drivers/phidgets_drivers #152: Humble Branch compiles issues

austin-InDro, 2022-07-21:
Hello, I believe with some of the newer humble updates there is now an issue with the decleration of parameters. They now require the typename to be explicity mentioned like this:
```c++
int serial_num =
        this->declare_parameter<int>("serial", -1);  // default open any device
```
instead of 
```c++
int serial_num =
        this->declare_parameter("serial", -1);  // default open any device
```

I've just recenetly had this issue when performing a docker build on the newest version of Humble. Can anyone else confirm this?

If so, I can create a pull request to fix the various phidget nodes.

mintar, 2022-07-21:
I don't have access to my PC, so I can't try it myself. If true, it's weird that a basic API like that changed in a released distro. @clalancette , could you have a look?

clalancette, 2022-07-21:
Hm, I'm not seeing this behavior.  If I checkout this repository to the `humble` branch, and then build this against humble, everything seems to compile as expected.  Can you give more information on exactly how you were building and what branch you were using?

austin-InDro, 2022-07-22:
Hmmmm I just did a re-pull from the humble branch and built with my Dockerfile and it's fine.... I forked the repo and made an edit to the spatial node, but I must of done something else to cause this issue.

Either way, everything is fine, sorry for the false alarm!

closed 2022-07-22 by austin-InDro