ros-drivers/phidgets_drivers #159: MOT0110_0: Spatial: Failed to open device: Timed Out

Assume-Zhan, 2023-02-02:
Like the issue happened before, but hardware is PhidgetSpatial Precision 3/3/3 MOT0110_0

\
**Phidgets Device**
- PhidgetSpatial Precision 3/3/3 MOT0110_0

\
**Host Machine**
- PC
- Ubuntu 20.04
- ROS Noetic

\
**Issue Description**
- Time out
Everytime I run ```roslaunch phidgets_spatial spatial.launch``` it always get a timeout issue
```
process[imu_manager-1]: started with pid [19128]
process[PhidgetsSpatialNodelet-2]: started with pid [19129]
process[ImuFilterNodelet-3]: started with pid [19130]
[ INFO] [1675307869.990269765]: Loading nodelet /PhidgetsSpatialNodelet of type phidgets_spatial/PhidgetsSpatialNodelet to manager imu_manager with the following remappings:
[ INFO] [1675307869.994965673]: waitForService: Service [/imu_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1675307870.035434642]: Loading nodelet /ImuFilterNodelet of type imu_filter_madgwick/ImuFilterNodelet to manager imu_manager with the following remappings:
[ INFO] [1675307870.038978812]: waitForService: Service [/imu_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1675307870.053339743]: Initializing nodelet with 2 worker threads.
[ INFO] [1675307870.061116774]: waitForService: Service [/imu_manager/load_nodelet] is now available.
[ INFO] [1675307870.082608610]: Initializing IMU Filter Nodelet
[ INFO] [1675307870.084520133]: waitForService: Service [/imu_manager/load_nodelet] is now available.
[ INFO] [1675307870.085496595]: Starting ImuFilter
[ INFO] [1675307870.110404430]: Using dt computed from message headers
[ INFO] [1675307870.111183992]: The gravity vector is kept in the IMU message.
[ INFO] [1675307870.120631814]: Imu filter gain set to 0.100000
[ INFO] [1675307870.120804778]: Gyro drift bias set to 0.000000
[ INFO] [1675307870.120835962]: Magnetometer bias values: 0.000000 0.000000 0.000000
[ INFO] [1675307870.164524970]: Initializing Phidgets SPATIAL Nodelet
[ INFO] [1675307870.165013733]: Starting Phidgets SPATIAL
[ INFO] [1675307870.165059200]: Opening spatial
[ INFO] [1675307870.170164054]: Connecting to Phidgets Spatial serial -1, hub port 0 ...
[ERROR] [1675307871.171380593]: Spatial: Failed to open device: Timed Out
[FATAL] [1675307871.171892936]: Failed to load nodelet '/PhidgetsSpatialNodelet` of type `phidgets_spatial/PhidgetsSpatialNodelet` to manager `imu_manager'
[PhidgetsSpatialNodelet-2] process has died [pid 19129, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load phidgets_spatial/PhidgetsSpatialNodelet imu_manager __name:=PhidgetsSpatialNodelet __log:=/home/assume/.ros/log/bb27ff20-a2a2-11ed-b901-0242ac110002/PhidgetsSpatialNodelet-2.log].
log file: /home/assume/.ros/log/bb27ff20-a2a2-11ed-b901-0242ac110002/PhidgetsSpatialNodelet-2*.log
```

- Additional info
When I run ```lsusb```, it has shown up
```
Bus 002 Device 011: ID 06c2:008d Phidgets Inc. (formerly GLAB) 
```

I've run the command in README to setup udev rules.

I've also installed the latest libphidget22, and run the helloworld example in [here](https://www.phidgets.com/?tier=3&catid=10&pcid=8&prodid=1205)
And it always output 5 times even I have plugin and unplugin once.
```
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Goodbye Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Goodbye Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Goodbye Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Goodbye Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
Goodbye Device PhidgetSpatial Precision 3/3/3, Serial Number: 700697
```

Did I miss any required steps?

mintar, 2023-02-02:
> Like the issue happened before, but hardware is PhidgetSpatial Precision 3/3/3 MOT0110_0

What issue?

> I've also installed the latest libphidget22, and run the helloworld example in [here](https://www.phidgets.com/?tier=3&catid=10&pcid=8&prodid=1205)

Could you try the helloworld example with `libphidget22-1.7.20210816` instead of the latest one? This version is what we're using, and it should already have support for the MOT0110_0. If the helloworld example doesn't work with our version of libphidget22, then maybe upgrading to the latest version will solve your problem.

kunal-mod, 2023-02-28:
I am facing the same issue. I have MOT0110_0 and the HelloWorld example from libphidget22-1.13 works. But the phidigets_drivers package uses the 1.7 version of libphidget. How should I upgrade the version of libphidget of the phidgets_drivers package to make it run. I am new to all this so a little help would be really appreciated, thanks.

clalancette, 2023-02-28:
In short, you need to update https://github.com/ros-drivers/phidgets_drivers/blob/noetic/libphidget22/CMakeLists.txt to point to the new version, and make any patches necessary to make it compile without warnings.  You can see an example PR (on the `rolling` branch) here: https://github.com/ros-drivers/phidgets_drivers/commit/9b679fa5ad2d91832de163398d239ffa74dceaea

kunal-mod, 2023-03-01:
Thanks @clalancette for your answer, After changing the libphidget version to the appropriate one in the CMakeLists.txt file and using the patch file for the mentioned PR the package finally worked.

clalancette, 2023-03-01:
> Thanks @clalancette for your answer, After changing the libphidget version to the appropriate one in the CMakeLists.txt file and using the patch file for the mentioned PR the package finally worked.

If you'd like to submit a PR to update noetic to that, it would be much appreciated.

closed 2023-03-02 by mintar

mintar, 2023-03-03:
I have merged the update to libphidget into the `noetic`, `rolling` and `humble` branches and released them on the ROS buildfarm. They should be part of the next binary release (in a few weeks time). Until then, you have to build from source.

andrespulido8, 2024-02-27:
I have the same issue. I just bought a PhidgetSpatial Precision 3/3/3 MOT0110_0. Ubuntu 20 and ROS noetic. 

I've tried to use the latest version of libphidget, `libphidget22-1.18.2024123`, and it worked well for the HelloWorld example and the `example.py` script to print the IMU measurements.
Later on, I tried to use the current library you are using for the noetic branch, `libphidget22-1.13.20230224`, and it can't detect any attach and detach signal.

I wanted to try to replicate the steps in for ex #161 but I am not sure where to get the `URL_MD5` value. I just change the library version, try to compile and I get a `Each Download failed!`. 

Could you instruct me how to do that so I can try to update the libphidget library my driver uses?

andrespulido8, 2024-02-27:
> I have the same issue. I just bought a PhidgetSpatial Precision 3/3/3 MOT0110_0. Ubuntu 20 and ROS noetic.
>
> I've tried to use the latest version of libphidget, `libphidget22-1.18.2024123`, and it [...]

I managed to get it using `md5sum path/to/libphidget22-1.18.2024123.tar.gz`. However, when compiling `libphidget22` with `catkin build` it never compiles, it just runs forever (>40 min), so I have not yet been able to use the new version of the library

mintar, 2024-02-28:
The patch file needs to be recreated as well. It probably hangs because: 

- The `patch` command doesn't succeed because you switched the library, so the `libphidget22-*.patch` doesn't match the files it is trying to patch.
- When the `patch` command fails, it goes into interactive mode and asks the user how to proceed.
- However, that query is swallowed by `catkin`, so you don't see it and it just hangs forever.

You need to compile it without the patch file, then you'll get a lot of warnings from `libphidget`. You have to fix those warnings and create a new patch file for those fixes.

andrespulido8, 2024-03-07:
Thanks @mintar. TLDR: I got it working. 

In case you are interested, after compiling without the patch file, I ran the spatial launch file again but I got the same `MOT0110_0: Spatial: Failed to open device: Timed Out` error. I tried many things; one was re-downloading the same library from the "User Guide" in the [sensor website](https://www.phidgets.com/?prodid=1205), then tried to run the "Example Code" with Python, which did not run so I installed the [python library](https://www.phidgets.com/docs/Language_-_Python) with `pip install --force-reinstall -v "Phidget22==1.18.20240123"`. After all that, I managed to get it working with the ros driver (although I am not sure exactly what made it work). 

Thanks again.

PR #161 (2023-03-02): Update to libphidget22 1.13
This is a backport of #160 to ROS noetic. 

Fixes #159.