ros-drivers/phidgets_drivers #162: Spatial: Failed to set algorithm magnetometer gain: Operation Not Supported

ammarazab1, 2023-03-14:
Hello,

When I run the spatial.launch with (use_orientation=true) this error occured

`[ INFO] [1678780967.616305822]: Loading nodelet /PhidgetsSpatialNodelet of type phidgets_spatial/PhidgetsSpatialNodelet to manager imu_manager with the following remappings:
[ INFO] [1678780967.617884863]: waitForService: Service [/imu_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1678780967.621521091]: Loading nodelet /ImuFilterNodelet of type imu_filter_madgwick/ImuFilterNodelet to manager imu_manager with the following remappings:
[ INFO] [1678780967.623138646]: waitForService: Service [/imu_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1678780967.640097989]: Initializing nodelet with 8 worker threads.
[ INFO] [1678780967.643929171]: waitForService: Service [/imu_manager/load_nodelet] is now available.
[ INFO] [1678780967.648452280]: Initializing IMU Filter Nodelet
[ INFO] [1678780967.649313458]: Starting ImuFilter
[ INFO] [1678780967.653332179]: Using dt computed from message headers
[ INFO] [1678780967.653355960]: The gravity vector is kept in the IMU message.
[ INFO] [1678780967.659930734]: waitForService: Service [/imu_manager/load_nodelet] is now available.
[ INFO] [1678780967.662224063]: Imu filter gain set to 0.100000
[ INFO] [1678780967.662249266]: Gyro drift bias set to 0.000000
[ INFO] [1678780967.662262572]: Magnetometer bias values: 0.000000 0.000000 0.000000
[ INFO] [1678780967.672481123]: Initializing Phidgets SPATIAL Nodelet
[ INFO] [1678780967.672569152]: Starting Phidgets SPATIAL
[ INFO] [1678780967.672585845]: Opening spatial
[ INFO] [1678780967.680755154]: Connecting to Phidgets Spatial serial 700822, hub port 0 ...
[ INFO] [1678780967.691228713]: Connected
[ INFO] [1678780967.692650948]: Calibrating IMU, this takes around 2 seconds to finish. Make sure that the device is not moved during this time.
[ INFO] [1678780969.693107665]: Calibrating IMU done.
[ERROR] [1678780969.694381734]: Spatial: Failed to set algorithm magnetometer gain: Operation Not Supported
[ INFO] [1678780969.940675591]: Phidget Spatial detached.
[FATAL] [1678780970.199846764]: Failed to load nodelet '/PhidgetsSpatialNodelet` of type `phidgets_spatial/PhidgetsSpatialNodelet` to manager `imu_manager'
[PhidgetsSpatialNodelet-3] process has died [pid 114660, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load phidgets_spatial/PhidgetsSpatialNodelet imu_manager __name:=PhidgetsSpatialNodelet __log:=/home/roboads/.ros/log/9b6ec6b2-c23e-11ed-9561-1b7d426659eb/PhidgetsSpatialNodelet-3.log].
log file: /home/roboads/.ros/log/9b6ec6b2-c23e-11ed-9561-1b7d426659eb/PhidgetsSpatialNodelet-3*.log
`

I use new package and phidgets22 library libphidget22-1.13.20230224
and this is spatial.launch file

`

<launch>

  #### Nodelet manager ######################################################

  <node pkg="nodelet" type="nodelet" name="imu_manager"
    args="manager" output="screen" />

  #### Spatial Driver #######################################################

  <node pkg="nodelet" type="nodelet" name="PhidgetsSpatialNodelet"
    args="load phidgets_spatial/PhidgetsSpatialNodelet imu_manager"
    output="screen">

    # See README.md for more information on the parameters

    # optional param serial, default is -1
    <param name="serial" value="700822"/>

    # optional param hub_port, used if connected to a VINT hub
    

    # optional param frame_id, default is "imu_link"
    

    # optional param use_orientation, default is false
    <param name="use_orientation" value="true"/>

    # optional param spatial_algorithm, default is "ahrs"
    

    # optional ahrs parameters
    

    # optional param algorithm_magnetometer_gain, default is 0.005
    

    # optional param heating_enabled, not modified by default
    

    # optional param linear_acceleration_stdev, default is 280ug
    

    # optional param angular_velocity_stdev, default is 0.095 deg/s
    

    # optional param magnetic_field_stdev, default is 1.1 milligauss
    

    # supported data rates: 4 8 16 24 32 40 ... 1000 (in ms)
    <param name="data_interval_ms" value="4"/>

    # optional param publish_rate, defaults to 0
    

    # optional param server_name, required to locate remote network servers
    

    # optional param server_ip, required to locate remote network servers
    

    # compass correction params (see http://www.phidgets.com/docs/1044_User_Guide)
    
  </node>

  #### IMU Orientation Filter ###############################################

  <node pkg="nodelet" type="nodelet" name="ImuFilterNodelet"
    args="load imu_filter_madgwick/ImuFilterNodelet imu_manager"
    output="screen">

    <param name="use_mag" value="false"/>
  </node>

</launch>
`

Thanks in advance

mintar, 2023-03-14:
What version of Phidgets Spatial do you have? From the README:

> * `use_orientation` (bool) - Use the phidget spatials onboard orientation estimation; Just available on MOT0109 onwards. Set to false for older versions. Defaults to false.

ammarazab1, 2023-03-14:
Ooh I doesn't read all parameter in Readme file so I using MOT0110_0 so it doesn't work

a last question is the orientation data output from imu filter is opposite to correct direction like this photo

[image]

mintar, 2023-03-14:
I'm not quite sure I understand your second question, but perhaps this video helps understand the orientations of the IMU. Note that the IMU has to be "upside down" for the z axis to point up:

https://user-images.githubusercontent.com/320188/224983807-79398d93-8366-40b1-8bf6-0e1055e75d9f.mp4

ammarazab1, 2023-03-14:
Thanks @mintar I rotate the IMU to be all axis similar to  rviz imu visualize and all thing going well 
Thanks and I will close the Issues.

closed 2023-03-14 by ammarazab1