ros-drivers/phidgets_drivers #163: API of MOT0109 and MOT0110 is different

Assume-Zhan, 2023-03-17:
Hello, 

I'm using the Phidgets driver spatial precision MOT0110, and I'm encountering an error when I use the "use_orientation" parameter. Specifically, I receive the following error message:
```
Spatial: Failed to set algorithm magnetometer gain: Operation Not Supported
```

Although this param is set before ```MOT0109``` in README, I still try to find whats going wrong : 

Even though the parameter is set before MOT0109 in the README, I'm still having issues. Upon further investigation, I discovered that the Phidget API uses an unsupported function for MOT0110, possibly at this line of code:
[HERE](https://github.com/ros-drivers/phidgets_drivers/blob/noetic/phidgets_api/src/spatial.cpp#L179)

It appears that the MOT0110 API does not have a ```PhidgetSpatial_setAlgorithmMagnetometerGain()``` function, whereas MOT0109 does.

@mintar, is there any way to resolve this issue? Or have I misunderstood something?

mintar, 2023-03-17:
AFAIK, the MOT0110 doesn't support onboard orientation estimation, so the solution is to set `use_orientation` to `false`.

Assume-Zhan, 2023-03-18:
Ok, thank you !

closed 2023-03-18 by Assume-Zhan