ros-drivers/phidgets_drivers #164: Drivers issues with older version of Phidgets

CAPlevel, 2023-03-21:
I have been using Phidgets for years.
I build a robot based on Phidgets & Flowbotics.
I decided to move to ROS2, and I saw as a blessing to have the drivers for Ros and Phidget.
unfortunately, since my boards are the older version and don't support some commands. most of the test I try to make launch an error since some functions are not supported, even do i don't need it for my project.
I updated the files to the latest, but every driver I tested, expect some of the new functions on my boards and abort.
For Example:

1064 Motor controller:

[component_container-1] [ERROR] [1679238490.289006300] 
[phidgets_motors]: Motors: Failed to set braking strength for Motor 
channel 0: Operation Not Supported.

Similar problems with 1018 Analog input:
[phidgets_analog_inputs]: AnalogInputs: Failed to set analog input voltage range: Operation Not Supported [component_container-1] [ERROR] [1679409297.142145890]

it seems to me this ROS driver does not support older Phidget 1064 and older Phidget 1018, I haven't try other boards.

Do you have a version of the ROS driver for older boards? Is frustrating having those drivers, I'll need to re-write all containers or create new node.

I appreciate your help and Advise if there is other workaround for this issues.

best regards,

Haroldo Level

************************************ LAUNCH COPY  of the 1064 MOTOR CONTROLLER ****************
aroldo@haroldo-desktop:~$ ros2 launch phidgets_motors motors-launch.py
[INFO] [launch]: All log files can be found below 
/home/haroldo/.ros/log/2023-03-19-11-08-07-878521-haroldo-desktop-3935
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [3949]
[component_container-1] [INFO] [1679238489.640349975] 
[phidget_container]: Load Library: /opt/ros/humble/lib/libphidgets_motors.so
[component_container-1] [INFO] [1679238489.712897581] 
[phidget_container]: Found class: 
rclcpp_components::NodeFactoryTemplate<phidgets::MotorsRosI>
[component_container-1] [INFO] [1679238489.713108207] 
[phidget_container]: Instantiate class: 
rclcpp_components::NodeFactoryTemplate<phidgets::MotorsRosI>
[component_container-1] [INFO] [1679238489.731637991] [phidgets_motors]: 
Starting Phidgets Motors
[component_container-1] [INFO] [1679238489.732044040] [phidgets_motors]: 
Connecting to Phidgets Motors serial -1, hub port 0 ...
[component_container-1] [INFO] [1679238490.275097290] [phidgets_motors]: 
Connected to serial 100346, 2 motors
[component_container-1] [INFO] [1679238490.288302534] [phidgets_motors]: 
Back EMF sensing not supported for motor_back_emf00
[component_container-1] [ERROR] [1679238490.289006300] 
[phidgets_motors]: Motors: Failed to set braking strength for Motor 
channel 0: Operation Not Supported
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 
'phidgets_motors' of type 'phidgets::MotorsRosI' in container 
'/phidget_container': Component constructor threw an exception: Failed 
to set braking strength for Motor channel 0: Operation Not Supported
[component_container-1] [ERROR] [1679238490.349020564] 
[phidget_container]: Component constructor threw an exception: Failed to 
set braking strength for Motor channel 0: Operation Not Supported
******************************

************************************ LAUNCH COPY  of the 1018 analog inputs board ****************

haroldo@haroldo-desktop:~$ ros2 launch phidgets_analog_inputs analog_inputs-launch.py [INFO] [launch]: All log files can be found below
/home/haroldo/.ros/log/2023-03-21-10-34-55-984324-haroldo-desktop-56745
[INFO] [launch]: Default logging verbosity is set to INFO [INFO] [component_container-1]: process started with pid [56759] [component_container-1] [INFO] [1679409296.833921271]
[phidget_container]: Load Library: 
/opt/ros/humble/lib/libphidgets_analog_inputs.so
[component_container-1] [INFO] [1679409296.852532703]
[phidget_container]: Found class: 
rclcpp_components::NodeFactoryTemplate<phidgets::AnalogInputsRosI>
[component_container-1] [INFO] [1679409296.852756201]
[phidget_container]: Instantiate class: 
rclcpp_components::NodeFactoryTemplate<phidgets::AnalogInputsRosI>
[component_container-1] [INFO] [1679409296.871127248]
[phidgets_analog_inputs]: Starting Phidgets AnalogInputs [component_container-1] [INFO] [1679409296.871527410]
[phidgets_analog_inputs]: Connecting to Phidgets AnalogInputs serial -1, hub port 0 ...
[component_container-1] [ERROR] [1679409297.130465748]
[phidgets_analog_inputs]: AnalogInputs: Failed to set analog input voltage range: Operation Not Supported [component_container-1] [ERROR] [1679409297.142145890]
[phidget_container]: Component constructor threw an exception: Failed to set analog input voltage range: Operation Not Supported [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_analog_inputs' of type 'phidgets::AnalogInputsRosI' in container '/phidget_container': Component constructor threw an
exception: Failed to set analog input voltage range: Operation Not Supported

^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) [component_container-1] [INFO] [1679409325.315882633] [rclcpp]: 
signal_handler(signum=2)
[ERROR] [component_container-1]: process has died [pid 56759, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container
--ros-args -r __node:=phidget_container -r __ns:=/'].
haroldo@haroldo-desktop:~$

clalancette, 2023-03-21:
In short, if these are optional things then I think it would be OK to catch the exceptions and continue on.  For instance, you could put a `try..catch` block around https://github.com/ros-drivers/phidgets_drivers/blob/6974eafd1315a7c958f1970ca44b26a622bdd14c/phidgets_motors/src/motors_ros_i.cpp#L118 , and handle the `Phidgets22Exception` that gets thrown, and go on.

CAPlevel, 2023-03-21:
Thanks for your fast response.
I can try that.
Do you have a link to help me how to compile the new files.cpp so it can be
used by the command?

Thanks

Haroldo Level

clalancette, 2023-03-21:
You can follow the general advice at http://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html to create a workspace with phidgets_drivers in it, and go from there.

closed 2023-11-14 by mintar