ros-drivers/phidgets_drivers #165: PhidgetManager_setOnErrorHandler symbol not found

Doomerdinger, 2023-06-22:
It appears that the function `PhidgetManager_setOnErrorHandler` is not present in the library file, though it is present in the phidgets header.
If it is ever used, you end up with an undefined reference error at compile time.

Two similar functions (`PhidgetManager_setOnAttachHandler` and `PhidgetManager_setOnDetachHandler`) seem to work fine.

In older versions of this library (melodic I believe) the function signatures were slightly different (`CPhidgetManager_set_OnAttach_Handler` for example) and all three seemed to work.

In the source code of the phidget22 library downloaded from phidget's website I see the function `PhidgetManager_setOnErrorHandler` present in the `phidget22.h` and `manager.h` headers, alongside the other two sister functions mentioned above.
Similarly, all are defined in the `manager.c` file.
The `PhidgetManager_setOnAttachHandler` function does seem to be absent from the `src/plat/linux/symversion.def` file where the other two are present, but I do not know what this indicates. 

I'm not overly familiar with this codebase, I apologize if this is an intentional removal or if that function shouldn't really be used.

clalancette, 2023-06-22:
As far as I can tell, we never call `PhidgetManager_setOnErrorHandler` (in point of fact, we don't call the `PhidgetManager_` set of APIs at all).

Note that this is the ROS wrapper for the `libphidgets` library; we don't do development on the `libphidgets` library itself.  I think you'd have to contact support via https://www.phidgets.com/ to file a bug report with them.

Doomerdinger, 2023-06-23:
Yeah, I just was not sure if the way the `libphidget22` package configures/builds the library was causing this. It does not _seem_ like it but I wanted to double check since my knownledge of this is fairly slim.
Assuming that is not the case I will close this and take it up with phidgets.

Doomerdinger, 2023-06-28:
I'll assume the issue lies with the vendor code.

closed 2023-06-28 by Doomerdinger