ros-drivers/phidgets_drivers #183: [ROS2] there are no estimated orientation

NR-Daichi, 2024-05-15:
`ros2 launch phidgets_spatial spatial-launch.py`

with the above command, it successfully fires up node and publish imu/data_raw, imu/is_calibrated, and imu/mag. However it does not publish imu/data topic. Moreover, it shows the following errors if I set spatial_algorithm as ahrs. So I have to set it as imu.

`ros2 launch phidgets_spatial spatial-launch.py 
[INFO] [launch]: All log files can be found below /home/nr_daichi/.ros/log/2024-05-15-15-12-43-587291-ronot-40085
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [40097]
[component_container-1] [INFO] [1715778763.910926202] [phidget_container]: Load Library: /home/nr_daichi/ros2_ws/install/phidgets_spatial/lib/libphidgets_spatial.so
[component_container-1] [INFO] [1715778763.912989127] [phidget_container]: Found class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1715778763.913002177] [phidget_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1715778763.914901170] [phidgets_spatial]: Starting Phidgets Spatial
[component_container-1] [INFO] [1715778763.915258380] [phidgets_spatial]: Connecting to Phidgets Spatial serial -1, hub port 0 ...
[component_container-1] [INFO] [1715778763.921194557] [phidgets_spatial]: Connected to serial 532468
[component_container-1] [INFO] [1715778763.922114487] [phidgets_spatial]: Calibrating IMU, this takes around 2 seconds to finish. Make sure that the device is not moved during this time.
[component_container-1] [INFO] [1715778765.922641139] [phidgets_spatial]: Calibrating IMU done.
[component_container-1] [ERROR] [1715778765.923026648] [phidgets_spatial]: Spatial: Failed to set spatial algorithm: Operation Not Supported
[component_container-1] [ERROR] [1715778765.928116129] [phidget_container]: Component constructor threw an exception: Failed to set spatial algorithm: Operation Not Supported
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container '/phidget_container': Component constructor threw an exception: Failed to set spatial algorithm: Operation Not Supported
`

But with ununtu 20.04, it publishes topic imu/data and it has estimated orientation

mintar, 2024-05-15:
What exact phidgets device are you using? What versions of this repo and libphidgets are you using in ROS2 and ROS1? What is your ROS2 version?

I guess that the reason for this bug is that your phidgets device is pretty new, and that your ROS1 setup uses a newer libphidgets than your ROS2 setup (which still has a libphidgets version that doesn't support your device yet).

I thought I had fixed this recently. I'm traveling right now, so I cannot check. Also there was a similar issue recently, just search the closed issues.

mintar, 2024-05-15:
Related issues: #162, #159

mintar, 2024-05-15:
One more tip: make sure that the /imu/data topic on Noetic is actually published directly by the phidgets_spatial node and not using the imu_filter_madgwick.

NR-Daichi, 2024-05-17:
It's my apologizes that I omitted information. The followings are my set up:

IMU: PhidgetSpatial 3/3/3 Basic 1042_0
ROS2: Humble with Ubuntu 22.04 + humble branch of this rep (https://github.com/ros-drivers/phidgets_drivers/tree/humble)
ROS1: Noetic with Ubuntu 20.04 + noetic branch of this rep

Additionally, I installed it from source. Is there any possibility that I am missing some dependencies?

mintar, 2024-05-18:
Your device (PhidgetSpatial 3/3/3 Basic 1042_0) doesn't support onboard orientation estimation. You can see that if you check the C API: https://www.phidgets.com/?prodid=1025#Tab_API

For your device, `PhidgetSpatial_getAlgorithm()` has a constant output of `Constant: SPATIAL_ALGORITHM_NONE`, and the function `PhidgetSpatial_setAlgorithm()` is missing. Compare that to the PhidgetSpatial Precision 3/3/3 MOT0110_0, which has that function: https://www.phidgets.com/?prodid=1205#Tab_API

This explains the error message that you're getting:

```
[component_container-1] [ERROR] [1715778765.923026648] [phidgets_spatial]: Spatial: Failed to set spatial algorithm: Operation Not Supported
```

However, you can still use your device! On ROS1 Noetic it works because it doesn't use the **onboard** orientation estimation. Instead it uses `imu_filter_madgwick` to estimate the orientation.

You can do the same on ROS2: https://github.com/CCNYRoboticsLab/imu_tools/tree/humble

NR-Daichi, 2024-05-21:
Thank you for your help.

closed 2026-09-17 by mintar