ros-drivers/phidgets_drivers #184: M0T0109_0: Spatial: Failed to open device: Timed Out / ROS 2 Humble

AnaelLaura98, 2025-03-05:
Hello I am working with a ROS Humble based Docker Container and I am trying to launch the spatial drivers node but I got this Error:

ki@ki-desktop:~$ docker logs phidgets-drivers 
sourcing   /opt/ros/humble/setup.bash
sourcing   /home/docker/ros2_ws/install/setup.bash
[INFO] [launch]: All log files can be found below /home/docker/.ros/log/2025-03-05-13-49-43-653438-399448fa61f5-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [62]
[component_container-1] [INFO] [1741182586.424084456] [phidget_container]: Load Library: /opt/ros/humble/lib/libphidgets_spatial.so
[component_container-1] [INFO] [1741182586.441169709] [phidget_container]: Found class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1741182586.453229134] [phidget_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1741182586.541753021] [phidgets_spatial]: Starting Phidgets Spatial
[component_container-1] [INFO] [1741182586.545234713] [phidgets_spatial]: Connecting to Phidgets Spatial serial -1, hub port 0 ...
[component_container-1] [ERROR] [1741182587.555202274] [phidgets_spatial]: Spatial: Failed to open device: Timed Out
[component_container-1] [ERROR] [1741182587.567987766] [phidget_container]: Component constructor threw an exception: Failed to open device: Timed Out
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'phidgets_spatial' of type 'phidgets::SpatialRosI' in container '/phidget_container': Component constructor threw an exception: Failed to open device: Timed Out

I saw in an issue from 2021 that there was a need for developper to add the changes to the noetic branch but since i am working with the ROS2 version of the package, could my problem be also related to it?

I have installed the drivers as a package in my dockerfile like this:
sudo apt-get install ros-<ros_distro>-phidgets-drivers.

Thank you for your help!

mintar, 2025-03-05:
The previous issues (#105, #159) occurred because at the time this repo included an old version of the libphidget library that didn't have support for the MOT0109_0 resp. MOT0110_0 device yet. We upgraded to a newer version of libphidget that supports M0T0109_0 (`libphidget22-1.7.20210816`). The ROS2 version has an even more recent version (`libphidget22-1.19.20240304`), so this should not be the cause of the problems.

I suspect the real cause is that our ROS2 driver doesn't have full access to the Phidget device from inside the Docker container. Maybe you need to run Docker with the `--privileged` flag, change the permissions on the phidget USB device, install the udev rules or other stuff like this. You can also download and run the example code from the phidget website. If the device works with that, we should also get it running with our ROS driver.

mintar, 2025-03-06:
> An I still dont know why t impossible for me to apply the udev rules on my Host System

Because you are not `root`. Try `sudo ...`.

Did you try the instructions [here](https://www.phidgets.com/docs/OS_-_Linux#Setting_UDev_Rules)?

AnaelLaura98, 2025-03-10:
Hello I tried the instructions but the curl command was not working with me but here is how I made it happen:
I first created the rules and after I have downloaded the driver with the following curl command:
ki@ki-desktop:~$ sudo curl -fsSL https://www.phidgets.com/gpgkey/pubring.gpg -o /usr/share/keyrings/phidgets.gpg
ki@ki-desktop:~$ echo "deb [signed-by=/usr/share/keyrings/phidgets.gpg] https://www.phidgets.com/debian focal main" | sudo tee /etc/apt/sources.list.d/phidgets.list
deb [signed-by=/usr/share/keyrings/phidgets.gpg] https://www.phidgets.com/debian focal main
ki@ki-desktop:~$ sudo apt update
ki@ki-desktop:~$ sudo apt install -y libphidget22 libphidget22-dev

And afterward the helloWorld example was able to work correctly:
ki@ki-desktop:~/Downloads/Phidget22_HelloWorld_C_Ex_20250106/HelloWorld_Example$ ./HelloWorld
Opening...
Phidget Simple Playground (plug and unplug managers)
Press Enter to end anytime...
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 373530
Hello Device PhidgetSpatial Precision 3/3/3, Serial Number: 373530

I have repeated the exact same process in my Dockerfile this way:

# Phidget GPG-Key herunterladen und Repository hinzufügen
RUN curl -fsSL https://www.phidgets.com/gpgkey/pubring.gpg -o /usr/share/keyrings/phidgets.gpg && \
    echo "deb [signed-by=/usr/share/keyrings/phidgets.gpg] https://www.phidgets.com/debian focal main" | tee /etc/apt/sources.list.d/phidgets.list

# Phidget-Treiber installieren
RUN apt update && apt install -y libphidget22 libphidget22-dev

# Udev-Regel für Phidgets-Geräte hinzufügen
RUN echo 'SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="06c2", ATTRS{idProduct}=="00[3-a][0-f]", MODE="666"' > /etc/udev/rules.d/99-libphidget22.rules

And it has worked and my driver run now in my container also:
docker@84792c698119:~/ros2_ws$ ros2 launch phidgets_spatial spatial-launch.py
[INFO] [launch]: All log files can be found below /home/docker/.ros/log/2025-03-10-12-13-05-170952-84792c698119-154
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [166]
[component_container-1] [INFO] [1741608785.629413110] [phidget_container]: Load Library: /opt/ros/humble/lib/libphidgets_spatial.so
[component_container-1] [INFO] [1741608785.633987873] [phidget_container]: Found class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1741608785.634174533] [phidget_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<phidgets::SpatialRosI>
[component_container-1] [INFO] [1741608785.648964760] [phidgets_spatial]: Starting Phidgets Spatial
[component_container-1] [INFO] [1741608785.654001739] [phidgets_spatial]: Connecting to Phidgets Spatial serial -1, hub port 0 ...
[component_container-1] [INFO] [1741608785.696094402] [phidgets_spatial]: Connected to serial 373530
[component_container-1] [INFO] [1741608785.702362921] [phidgets_spatial]: Calibrating IMU, this takes around 2 seconds to finish. Make sure that the device is not moved during this time.
[component_container-1] [INFO] [1741608787.706416156] [phidgets_spatial]: Calibrating IMU done.
[component_container-1] [INFO] [1741608787.706766370] [phidgets_spatial]: No compass correction params found.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/phidgets_spatial' in container '/phidget_container'

Thank you so much for the help.

I hope this will help other person who will encouter the same Problem.

closed 2025-03-11 by mintar