ros-drivers/phidgets_drivers #193: Mounting Advice for ROS

mfruhner, 2025-06-16:
Hey,
I have a question regarding the mounting orientation of a PhdigetSpatial Precision 3/3/3 on a mobile robot.
We have a four-wheeled rectangularly shaped robot platform and want to mount the IMU in the center.

I which orientation should the IMU be mounted so that i complies with the ROS convention of X-forward, Y-left, Z-up?
As I understand from the documentation, the X axis goes from left to right, Y from front to back, w.r.t. the USB Port.

Can we mount it in any direction and apply a matching static transform (which one?) or should we place it in the correct orientation (which one?) from the beginning?

Thanks for your advice!

mintar, 2025-06-16:
On my old PhidgetSpatial 1044 3/3/3, the axes are printed onto the board; IIRC, axis 0=x, 1=y, 2=z. You can easily figure this out (or verify it) yourself for your IMU. Just install this driver, turn the IMU onto each of the 6 sides and echo the raw IMU values published by the driver. If the linear acceleration for axis x is positive and roughly +9.81 N, and the linear acceleration for y and z are roughly 0, then this is the direction of the positive x axis. Same for the y and z axes.

The only thing that really matters is the direction of the z axis. The ROS standard is to use the ENU convention for IMUs (x/y/z = East/North/Up), so the z axis should point up. For my IMU, this means mounting it "upside down" (i.e., with the underside of the PCB pointing up). Some packages, like `robot_localization`, [assume the IMU data to follow the ENU convention](https://docs.ros.org/en/noetic/api/robot_localization/html/preparing_sensor_data.html#coordinate-frames-and-transforming-sensor-data). However, if you mount your IMU in a different orientation, you should be able to transform it using [`imu_transformer`](https://github.com/ros-perception/imu_pipeline).

mfruhner, 2025-06-17:
Thank you for this detailed answer!
Could you please tell me, why only the Z axis matter?
The Axis are printed on the devices exactly as you say, which means that with the USB Port of the device pointing to the rear of the robot, Y points forward, X points left and Z points down.
Rolled upside down, X points right. Don't I need to rotate again around Z to make X point forward?

Thank you for your time!

mintar, 2025-06-18:
If you want, you can make X point forward. This has the advantage that if you look at the `linear_acceleration` field and `angular_velocity` of the `Imu` message, they follow ROS standard (X forward, Y left, Z up).

What I meant by "only Z matters" is that in my use case, I almost never look at the Imu message directly. I feed it to a filter (in my case `imu_filter_madgwick`) that fills the `orientation` field. (NB: If you have a newer Phidgets, you can set the `use_orientation` parameter to `true` so you don't need an external filter, see [README](https://github.com/ros-drivers/phidgets_drivers/blob/noetic/phidgets_spatial/README.md)).

The orientation field reports the orientation of the IMU with respect to a reference frame. That reference frame is ENU if you enable the magnetometer. However, I never enable the magnetometer because on small robots, the disturbance from the electric motors is too large. Without a magnetometer, the orientation of the x and y axes of the reference frame is arbitrary (it's just the initial orientation of the axes when you started the filter) because there is no reference measurement for these axes; the Z axis is special because the gravity vector gives us a reference measurement that tells us which way the Z axis should point. This is what I meant by "only Z matters", which was not 100% correct.

BTW, later stages in the pipeline like `robot_localization` and `amcl` don't care about the relative orientation of the x and y axes anyway, they only look at the relative change in angles.

closed 2026-01-16 by mfruhner